pi-top/Device-Management #6: Installation of pt-device-manager on latest Raspbian Stretch image is broken

rricharz, 2017-10-31:
pt-device-manager is now a known package in Raspbian Stretch, but it does not install in the latest Raspbian Stretch image due to a dependency problem:

>  *sudo apt install pt-speaker
  Reading package lists... Done
  Building dependency tree       
  Reading state information... Done
  Some packages could not be installed. This may mean that you have
  requested an impossible situation or if you are using the unstable
  distribution that some required packages have not yet been created
  or been moved out of Incoming.
  The following information may help to resolve the situation:
  The following packages have unmet dependencies:
   pt-speaker : Depends: pt-device-manager (>= 1.0.1) but it is not going to be installed
  E: Unable to correct problems, you have held broken packages.`

I traced it down to python3-pt-idletime not beig installable:

> *sudo apt install python3-pt-idletime
  Reading package lists... Done
  Building dependency tree       
  Reading state information... Done
  Some packages could not be installed. This may mean that you have
  requested an impossible situation or if you are using the unstable
  distribution that some required packages have not yet been created
  or been moved out of Incoming.
  The following information may help to resolve the situation:
  The following packages have unmet dependencies:
   python3-pt-idletime : Depends: python3 (< 3.5) but 3.5.3-1 is to be installed
  E: Unable to correct problems, you have held broken packages.

m-roberts, 2017-10-31:
Hi Rene, I apologise for the teething issues - part of us testing a new packaging and deployment system!
I'm not sure what's causing this error specifically - I will see if we can get that resolved quickly.

rricharz, 2017-11-01:
The installation on Raspbian Stretch runs now smoothly, but the device manager crashes with the following log:
>Nov 01 17:26:07 pitop systemd[1]: Started pi-top device auto-detection and configuration daemon.
Nov 01 17:26:07 pitop pt-device-manager[1543]: Traceback (most recent call last):
Nov 01 17:26:07 pitop pt-device-manager[1543]:   File "/usr/lib/pt-device-manager/pt-device-manager", line 17, in <module>
Nov 01 17:26:07 pitop pt-device-manager[1543]:     from ptdm_controller import Controller
Nov 01 17:26:07 pitop pt-device-manager[1543]:   File "/usr/lib/pt-device-manager/ptdm_controller.py", line 6, in <module>
Nov 01 17:26:07 pitop pt-device-manager[1543]:     from ptdm_idle_monitor import IdleMonitor
Nov 01 17:26:07 pitop pt-device-manager[1543]:   File "/usr/lib/pt-device-manager/ptdm_idle_monitor.py", line 2, in <module>
Nov 01 17:26:07 pitop pt-device-manager[1543]:     from idletime import get_idle_time
Nov 01 17:26:07 pitop pt-device-manager[1543]: ImportError: No module named 'idletime'
Nov 01 17:26:08 pitop systemd[1]: pt-device-manager.service: Main process exited, code=exited, status=1/FAILURE
Nov 01 17:26:08 pitop systemd[1]: pt-device-manager.service: Unit entered failed state.
Nov 01 17:26:08 pitop systemd[1]: pt-device-manager.service: Failed with result xit-code'.
Nov 01 17:26:08 pitop systemd[1]: pt-device-manager.service: Service hold-off time over, scheduling restart.

python3-pt-idletime was installed by _sudo apt install pt-speaker_ as the following output shows:

>sudo apt install python3-pt-idletime
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-pt-idletime is already the newest version (1.0.0-2).
python3-pt-idletime set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

rricharz, 2017-11-01:
Found idletime.cpython-34m-arm-linux-gnueabihf.so in /usr/lib/python3.4/dist-packages. Nothing called *idletime* in /usr/lib/python3.5. There is nothing else in /usr/lib/python3.4. I think that the package is targeting the wrong python version.

m-roberts, 2017-11-02:
Hi Rene - we are submitting a revised package now. Please let me know how you get on, so that we can hopefully close this issue.

rricharz, 2017-11-03:
Installs and runs now properly on Raspbian Stretch. Issue can be closed.

closed 2017-11-07 by m-roberts