pi-top/Device-Management #8: Add machine-readable output to pt-battery

m-roberts, 2017-11-03:
This will simplify usage for external tools. Likely will be accessible with a command line parameter.
e.g.
```
pt-battery charging
pt-battery capacity
pt-battery time
pt-battery wattage
```
to just return a value, and not have to parse text

spl237, 2017-11-06:
The above works for me; a few refinements I would like to suggest.

Capacity as a percentage is more useful than as a figure in mAh - not sure which you intended. (Or is "wattage" the mAh value?)

Assuming "charging" returns a value to tell you if a battery is charging or not, it would also be useful to know whether the battery was discharging or connected to external power. This could conceivably be done by comparing charging and time, so charging = 1 and time = non-zero is battery charging, charging = 0 and time = non-zero is battery discharging, and time = 0 is battery neither charging nor discharging, which could equate to on external power. It's a bit kludgy - the alternative would be for charging to return 0 for not charging, 1 for charging, 2 for on external power or similar.

m-roberts, 2017-11-07:
Capacity is as a percentage, and wattage is the mAh value.
Currently, the functionality is as follows: 

Charging State: 0 is default; updates to 1 if charging, 0 if discharging
Capacity: 50 is default if battery is not found; unit: %
Time Remaining: 0 is default; unit: minutes
Wattage: 0 is wattage; unit: mAh

I propose that all default values are -1. As far as I can tell, this ought to give you all of the functionality that you are asking for. If by 'external power' you mean 'pi-top battery not connected', charging state can therefore be -1 (as opposed to 0) in this situation.

It is worth noting that 'time remaining' represents 'time to full' if charging, and 'time to empty' if discharging. I realise that this is unclear, so if you would prefer I can look into providing 'time to full' and 'time to empty' as separate values to be read.

rricharz, 2017-11-07:
Actually, there are 4 possible states:
Battery not present, battery charging, battery discharging, battery full

Please let me know if you change the the output of the standard command without arguments.

-1 is a good value for the defaults.

Also, on my pi-top v1 laptop, wattage is always 0. This is probably a bug.

spl237, 2017-11-07:
"External power" is battery present but fully charged, and with power connected so not discharging.

"Time remaining" as a single value is fine - that's how the ACPI values I read from laptop batteries seem to work.

m-roberts, 2017-11-07:
@rricharz - I will conduct a full review on all pi-top hardware after I have redesigned the specification of the software, and run it by both of you. The output won't change in terms of parsing beyond default values changing.
@spl237 - I will add 'battery full' state, then.

m-roberts, 2017-11-28:
See [here](https://github.com/pi-top/Device-Management/tree/feature/machineReadableOutput) for my proposed implementation for this.
@rricharz @spl237 - what do you think?

rricharz, 2017-11-29:
This is fine. Somebody will need to check that calling pt-battery multiple times to get a set of battery parameters ever few seconds does not cause too much cpu workload or a bottleneck on the i2c bus. Currently one call every 5 seconds is fine, except that it fills the log of the device manager with unnecessary information.

m-roberts, 2017-11-29:
High verbosity in the journal logging has been helpful in catching unexpected issues (given how new the software is), and was considered not much of a serious issue as the journal is space-limited. However, if you have a suggestion for reducing journal printing in this use case that you think is appropriate, please let me know.

This new system should work without bottlenecks, as it will use the device manager's cached value of the battery state (which is internally polling 4 times a second I think).

I have made initial changes to hub code to work with this new system, and intend to test this new functionality in the coming days.

spl237, 2017-11-30:
Looks pretty good. I assume charging_state 2 (full battery) is "full battery with mains power connected", and that any use of the device with no mains power will return charging_state 0?

Does it print the result to stdout, or does it return it as a return code from the program? It's not obvious to me which it is - I assume from the use of "print" it is the former. It would make life easier from my point of view if it could be a return code, but I can work with stdout as well, so not a huge problem.

m-roberts, 2017-11-30:
It does print to stdout, but it should be trivial for me to implement an exit code for charging state. I will look into that for you!

spl237, 2017-11-30:
Sorry - I meant to return values from all options as exit codes, not just charging states - as they are all numeric values, that ought to be do-able. It ought to be consistent - either return values as exit codes for all queried values, or return all values as text on stdout.

If you're not happy about returning exit codes of values up to several thousand (wattages), then don't worry about it - I can parse stdout easily enough!

m-roberts, 2017-11-30:
Okay, let's stick to stdout. I think it's more in line with the convention used for our other code.

spl237, 2017-12-04:
I've just written a modification to my plugin which looks to see if pt-battery is available on the system and uses it if it is; if not it drops back to attempting raw I2C access. This seems to work well in both circumstances, so in the absence of any problems being discovered, I'll use this when the machine-readable version of pt-battery is available.

FYI, there are a couple of bugs in the version of pt-battery on this branch; you are reading no_of_args_set before initialising it; I added "no_of_args_set = 0" at line 51 to fix this.

Also, all of args.charging_state, args.capacity etc are either True or False in print_battery_state_message - they are never None by this point, so you need to remove the checks on the arguments being None in this function. Otherwise, the function only ever prints charging state, irrespective of argument, because the first "or" is always true.

With those fixed, this works well with my plugin.

spl237, 2017-12-04:
One other thing that occurs to me - with the plugin like this, I need to make three separate calls to it, to read state, capacity and time, and each of these requires opening a shell, parsing the response etc, which is quite inefficient. Given the response is always being returned to stdout rather than as an exit code, it would be more efficient to have one option return all the data in one go, separated by spaces / commas / whatever. So something like "pt-battery -a" could return "0 76 123 250", where 0 is the status, 76 is the capacity, 123 is the time, 250 is the current. I think that would actually be a lot more useful than individual calls for each parameter.

m-roberts, 2017-12-04:
What about, for example:
```
$ pt-battery -s
1
$ pt-battery -c
53
$ pt-battery -t
312
$ pt-battery -sc
1 53
$ pt-battery -sct
1 53 312
```

spl237, 2017-12-04:
Yes, that'd be absolutely fine.

m-roberts, 2017-12-05:
@spl237,

[This](https://github.com/pi-top/Device-Management/commit/3ad4b83bd08380b83eee4a26ac06c918634e8365) commit should be what you're after.

Please let me know what your thoughts are. @rricharz, I have deliberately tried to ensure that this modification has no effect on anything pre-existing.

However, this does not yet account for new default values, as this requires further testing. There is a chance that this might cause some unexpected behaviour, though I'm hopeful that it should be a straightforward change.

rricharz, 2017-12-05:
The original output is OK, but -s does not produce any output -t, -c and -t are OK.

m-roberts, 2017-12-05:
D'oh! Old code. I'll get on that in a sec.

rricharz, 2017-12-06:
Everything OK now from my point of view.

spl237, 2017-12-06:
It would make life easier if all the output was on one line - saves parsing multiple lines from stdout. Trivial patch attached.
[oneline-patch.txt](https://github.com/pi-top/Device-Management/files/1535007/oneline-patch.txt)

spl237, 2017-12-06:
However, with this now all working and my plugin modified to work with it, there is a problem...

With my old plugin which talked directly to the I2C, CPU load as a result of the measurements was unnoticeable. But using the pt-battery command adds a very significant chunk of CPU when a measurement is taken - presumably because python is having to be started each time to parse the pt-battery script. The measurements are actually visible on the CPU monitor plugin as regular peaks - CPU load for each measurement peaks at around 10% of total, which is around 50% of a core. I don't think this is acceptable for what ought to be a background process - people will notice and complain!

Is there any way to get this overhead down? Off the top of my head I can't think of one.

rricharz, 2017-12-06:
I have looked into the cpu usage a few days ago. If it would be a python script, one could talk directly to the device manager to reduce the overhead a bit. My script https://github.com/rricharz/pi-top-pulse-battery-gauge does produce less if any noticeable spikes, unless the battery charge changes. But I am not aware of an easy way to talk to the device manager using C. And we would give up a lot of flexibility. Potentially one could also hook up to the publish server of the device manager.

I think therefore that this is the price to pay for the added flexibility.

spl237, 2017-12-06:
IMO, this level of overhead isn't acceptable for a battery gauge. My I2C version has no noticeable effect on the CPU loading; unless this overhead can be reduced to the point where it isn't noticeable on the CPU plugin, I think I'm going to have to access the battery in pitop v2 via I2C directly as well (assuming that is still possible?)

m-roberts, 2017-12-06:
I agree that this is too large of an overhead for a battery gauge.

@rricharz: I am unsure how your script is able to produce fewer spikes, given that the actual communication with the device manager is the same. I would be interested in trying to track down what the difference is, if this is the case.

The alternative is to attach to the request server and ping it directly without invoking Python, or use the request server to get initial values and then trigger an update when new values are received. 

Obviously, this is a bit more work. However, this would keep everything in C; no system calls, no Python.

m-roberts, 2017-12-06:
@benp44 has just made a good point, that the request server may not even be necessary: the publish server should update every few seconds with new values. If you cannot connect to it, you can reasonably assume that it is because the device manager is not running.

spl237, 2017-12-06:
That sounds fine - can someone please point me at some example C code for connecting to the publish server?

Also, does this work the same way on pi-top 1 and pi-top 2? I only have a pi-top 1 for testing (unless someone would like to send me a pi-top 2... ;) )

m-roberts, 2017-12-08:
Hi Simon,

The new pi-top has a new unified interface for everything. Whilst we are yet to document this in any substantial way, you can observe how it works by looking at `python3-pt-hub-v2`, hosted on our repository (apt.pi-top.com) to see how interfacing works.

rricharz, 2017-12-09:
@slp237
The code for the proof of concept for directly getting the battery information from the request server is at https://github.com/rricharz/poc-get-battery-info
No visible overhead on the CPU Usage Monitor Plugin

rricharz, 2017-12-09:
I have now put a fully operational example of a battery widget written in C and talking directly to the request server at the repository https://github.com/rricharz/pi-top-battery-widget.

m-roberts, 2017-12-13:
I think that the scope of this issue has grown beyond machine-readable output of pt-battery.
I would therefore suggest that we move this discussion to the https://github.com/raspberrypi-ui/lxplug-ptbatt repo.

In the meantime, I have begun to merge in the code changes for this issue. Everything seems to be functional from this repository, and I have completed most of the testing for the v2. I will move onto the codebase for the v1 (particularly looking to add 'externally powered' to charging state, which is done for the v2).

closed 2018-02-06 by m-roberts