pi-top/Device-Management #9: Fix pt-input for Raspbian Stretch

rricharz, 2017-11-09:
pt-input does currently not work on Raspbian Stretch. This is a libicui18n version problem. See journalctl.

m-roberts, 2017-11-09:
Thank you, Rene - I had identified this and forgotten about it.
Much like with `python3-pt-idletime`, I will write a quick fix for this until it can be addressed more thoroughly.

m-roberts, 2017-11-09:
Workaround: add Jessie repo to sources, install `libicu52` package.

    sudo sed -i "s/stretch/jessie/g" /etc/apt/sources.list
    sudo apt update
    sudo apt install -y libicu52
    sudo sed -i "s/jessie/stretch/g" /etc/apt/sources.list
    sudo apt update

rricharz, 2017-11-09:
I can confirm that the the brightness keys on my pi-top rev 1 laptop work properly in Raspbian Stretch if the above workaround proposed by m-roberts has been used to install **libicu52**, and **pt-hub** and **pt-input** are installed the standard way fromtheir Stretch repo. Also, it is possible to modify /etc/pi-top/pt-input/keyboard-commands to modify shortcut keys. The keycode for any key on any keyboard can be obtained using xev.

I recommend to leave the issue open until a more permanent solution has been implemented.

m-roberts, 2018-03-13:
I have closed this, as pt-input has been replaced for Stretch.

closed 2018-03-13 by m-roberts