pi-top/Device-Management #10: Installing pt-hub in Raspbian Stretch breaks desktop preferences

rricharz, 2017-11-14:
After installing pt-hub in Raspbian Stretch, it is not possible anymore to change the desktop font size anymore permanently. One can use Desktop Preferences, system tab, to change for example the font size from 12 to 10. This does in fact change the font size, but after the next reboot it is set to 12 again.

I think this has to do with pt-desktop being a dependency installed automatically by pt-hub. Users will not like that the behavior of their desktop is modified or broken if they just want to install "software drivers" for their pi-top hardware.

rricharz, 2017-11-14:
A simple workaround is to issue the following command after installing pt-hub:
`sudo systemctl disable pt-display'

m-roberts, 2018-01-11:
@rricharz I have released an updated pt-desktop package, whereby the UI changing components have been moved to a pt-display package. Therefore, using this updated package will not introduce modifications to desktop preferences so I will close this issue.

closed 2018-01-11 by m-roberts