pi-top/Device-Management #19: pt-idletime warning on Stretch

rricharz, 2017-12-14:
pt-idletime produces the following warnings in the log of pt-device-manager (see issue 2 on support.pi-top.com/support/discussions/topics/24000005154, reported by Mike Spivey)

Dec 14 11:54:17 pitop2 pt-device-manager[373]: Starting idle time monitor...
Dec 14 11:54:17 pitop2 pt-device-manager[373]: No protocol specified
Dec 14 11:54:17 pitop2 pt-device-manager[373]: Responder server ready.
Dec 14 11:54:17 pitop2 pt-device-manager[373]: No protocol specified
Dec 14 11:54:17 pitop2 pt-device-manager[373]: pt-idletime.get_idle_time() returned -1. Check the configuration of xhost.
Dec 14 11:54:18 pitop2 pt-device-manager[373]: Listening for requests...

But the screen blanking works properly. This warning is only produced once during boot.

m-roberts, 2017-12-14:
This is timing-related to the `pt-xhost-fix.service` systemd service which needs to be run on startup so that it can work. This specific issue is therefore low priority.

m-roberts, 2018-05-17:
@rricharz - I have now managed to migrate this out of a systemd service.
pt-device-manager's `postinst` script now runs the same command that `pt-xhost-fix` was doing.

The following line is modified in `/etc/lightdm/lightdm.conf`:

    display-setup-script=/usr/bin/xhost local:root

From the next boot, this setting is always correctly configured, and no warnings are added to the journal/syslog.

This will be released soon.

closed 2018-05-21 by m-roberts