pi-top/Device-Management #20: pt-devices sometimes misidentifies host device

m-roberts, 2018-01-22:
If it is not enabled, it will think that a v2 pi-top is a pi-topCEED due to I2C not being enabled.
Proposed solution: check for I2C before doing ID. pi-topCEED should not be identified simply by *not* having I2C detected.

m-roberts, 2018-02-05:
https://github.com/pi-top/Device-Management/commit/b88bd56e5bc006dcc20749f8a81bfc5359e61779

m-roberts, 2018-02-05:
Needs testing; `initialise` might not be the best place for this

m-roberts, 2018-02-05:
https://github.com/pi-top/pt-device-manager-deb/commit/a05d55d8afd24bd490936f8af2bd5d77efb691d2

m-roberts, 2018-02-06:
Now using python3-pt-common for getting and setting I2C
https://github.com/pi-top/Device-Management/commit/afa12aba10007118e7aa9b1f34c5a9312f4fc7a5

m-roberts, 2018-02-06:
@Eliz4567 This is now ready for testing.

Note to all: if this passes testing, [this PR](https://github.com/pi-top/Device-Management/pull/21) needs to be closed.

Eliz4567, 2018-02-07:
QA test - failed.

Tested on a pi-topV2. System still states that the host device is a pi-topCEED.
@m-roberts

m-roberts, 2018-02-07:
@Eliz4567 This test was primarily to ensure that new users will get the correct data even if I2C was initially switched off. What has been identified here appears to only be a problem when I2C is disabled while the device manager is running. I have revised the steps to disable the device manager before modifying the I2C state, as this better reflects the system state we are targeting.

I will create a separate issue for disabling I2C while the device manager is running.

Eliz4567, 2018-02-07:
QA test - passed

closed 2018-02-07 by Eliz4567