pi-top/Device-Management #27: pt-battery shows wrong state on rev 1 pi-top

rricharz, 2018-05-18:
After the latest update of pt-device-manager in standard Raspbian today, my pi-top rev 1 shows the wrong battery state during discharging:

pi@pitop1:~ $ pt-battery
Charging State: 1
Capacity: 88
Time Remaining: 1059
Wattage: -1
pi@pitop1:~ $

The Raspbian battery icon shows therefore charging instead of discharging.

My own tool shows that the battery pack  still delivers the proper information:

Status registers:
[0xA] Current                    : -230 mA
[0xD] Relative state of charge   : 88 %
[0xC] Max err of state of charge : 1 %
[0xE] Absolute state of charge   : 81 %
[0x19] Design Voltage            : 14800 mV
[0x3F] Cell voltage 1            : 4026 mV
[0x3E] Cell voltage 2            : 4012 mV
[0x3D] Cell voltage 3            : 4018 mV
[0x3C] Cell voltage 4            : 4015 mV
[0x9] Total voltage              : 16073 mV
[0x8] Temperature                : 24.6 C

This is therefore very likely a new software bug. It worked properly before installing the new pt-device-manager.

m-roberts, 2018-05-20:
There was some code change around this, though I thought that I had done full functional and regression testing on this. I will address this as a priority and try to release a patch soon

m-roberts, 2018-05-21:
Seems to be that the library is not converting the I2C value that is read from the register into 2's complement, and so a positive/negative value check is always positive. I am testing around this to ensure that adding this logic does not affect any other areas.

m-roberts, 2018-05-21:
I have submitted what I hope to be a fix to this to Raspberry Pi. Please let me know if you continue to experience issues after this package is live.

closed 2018-05-21 by m-roberts