pi-top/Device-Management #34: pt-device-manager install in Raspbian Lite

rricharz, 2018-08-30:
A while ago it was possible to install pt-device-manager in Raspbian Lite to be able to use pt-battery and pt-brightness. Using the latest image of Raspbian Lite and installing the latest version of pt-device-manager, a desktop is now installed onto Raspbian Lite and is called upon bootup. It does not appear to be possible to log in at that time or to get a console working.

A work-around is to use 'sudo raspi-config' immediately after installing pt-device-manager, and to set the bootup options to Console before booting the first time. If ssh is enabled, this can of course also be done remotely.

It is my strong opinion that it should be possible to install the 'pi-top hardware drivers', which means pt-device-manager, without any unnecessary or unwanted additional software such as a desktop.

I guess that this problem also applies when installing pt-device-manager on the latest image of Retropie, but have not tested this myself.

m-roberts, 2018-08-31:
Ah, this is not a use-case that I had anticipated when I updated the package.

pt-device-manager recommends `lightdm`, as it is responsible for running `/usr/bin/xhost local:root`, as explained [here](https://github.com/pi-top/Device-Management/#user-input).

I also didn't realise that simply installing lightdm would cause a boot to desktop. I will therefore look into this and get back to you.

As ever, Rene, thank you!

m-roberts, 2018-09-10:
I was thinking about this today, and I think that this should work:
`sudo apt install --no-recommends pt-devices`

Lightdm is a recommendation of pt-device-manager, which itself is a dependency of `pt-devices` through other packages. By only installing the required dependencies of `pt-devices`, this should be skipped without issue.

I am a bit too busy to be able to try this in the near future - if you are able to try this and let me know, I would appreciate it! I will then add it to the README.

rricharz, 2018-09-27:
--no-recommends does not appear to get accepted as an option

m-roberts, 2018-09-28:
`sudo apt-get install --no-recommends pt-devices`?

rricharz, 2018-09-28:
--no-install-recommends is recognized, but gives the following error during install

Collecting python-osc
Downloading https://files.pythonhosted.org/packages/9c/2c/.....
Complete output from command python setup.py egg_info:
Traceback (most recent call last):
File "<string>", line 1, in <module>
Import error: No module named 'setuptools'

pt-battery gives after a restart:
Error getting battery info:  Resource temporarily unavailable

m-roberts, 2018-10-04:
Okay - sorry for the delay. I will take a look at this shortly!

odo2063, 2019-08-23:
Hi!

If you install without automated dependencies you need to install all but the graphical stuff by hand.

Hope that (still) helps...

rricharz, 2019-08-24:
I propose to close this issue as there is no easy solution and it does not appear to be very important.