pi-top/Device-Management #36: poweroff-v2 does not work

marcin-sielski, 2018-12-03:
Hi,

poweroff-v2 setups pi-tophub in some state (green led is blinking) but it does not cause the pi-tophub to power off even there is no HDMI output (hdmi_blanking=2, vcgencmd display_power 0)

Best Regards

Marcin Sielski

m-roberts, 2018-12-03:
Hi - we have not had reports of other hubs behaving like this, and it is not something that we are able to reproduce.

Would you mind providing more information?
Are you running on Raspbian or pi-topOS?
What happens when you run the service yourself?

marcin-sielski, 2018-12-03:
Hi,

Thank you for the response. I have already provided the logs in the pi-top support system ticket 22039 but I get no response ;(.

I have tested Raspbian and pi-topOS. After fresh SW install reboot and poweroff rarely worked correctly (did not worked most of the time). I have then started to investigate the issue myself: try to play with various kernel boot options etc.

Currently I came to the point with the configuration where reboot works just fine, power off does not work at all. When I trigger poweroff-v2 manually I get green blinking led an nothing happens. Once I have experienced the situation where I have run the script manually (green led was blinking) and I have left the pi-top for a longer period untouched (display went off - assuming some kind of power saver mechanism) and then pi-top powered off correctly. 

Thank you for the help in advance.

Best Regards

Marcin Sielski

toby-pi-top, 2019-07-01:
@m-roberts