pi-top/Device-Management #43: Pitop Rasp4 and RaspOS not working

pikowski, 2020-09-18:
Hallo,
i have problem running pt-device-manager on my PITOP, now running with a Raspberry Pi 4 and latest Raspberry Pi OS.
Former Raspbian is running well on the same hardware once i am starting from my sd-card with Raspbian. Command pt-devices gives back: Unknown host device. No change of brightness possible and no battery status there. Is there anything what i can do?

artosa, 2020-10-21:
I have same issue. I've got Pi-Top DIY box set up with Raspberry Pi 4 and latest Raspberry Pi OS.

Taking a look at the code: It's getting somehow device id of -1, which of course does not map against the known devices. So it seems something is not working in that initial device id query.

Similarly: No LCD, no brightness, battery status available. Perhaps Raspberry requires some module or setting?

m-roberts, 2020-10-21:
Hi - at time of writing, this repo and the packages provided to the Raspberry Pi apt repository are quite out of date.

We are in the process of rewriting our packages so that they are more easily installed onto Raspberry Pi OS. For the foreseeable future, it will be necessary to add our apt repository to install the packages.

To add the pi-topOS apt repo, please do the following:
```
sudo apt-key adv --fetch-keys https://apt.pi-top.com/pt-apt.asc
echo "deb https://apt.pi-top.com/pi-top-os sirius main contrib non-free" | sudo tee /etc/apt/sources.list.d/pi-top.list &> /dev/null
sudo apt update
```

In the meantime, if you have any issues, we ask that you use pi-topOS, which is almost identical to Raspberry Pi OS under the hood.

artosa, 2020-10-21:
Oh yes, that did the trick beautifully, now everything makes sense here. Thank you for quick reply.